Xilinx/PYNQ #882: _video._capture initialization failed [1]: No incoming video signal detected

chesterkyles, 2019-08-03:
Using PYNQ API v2.0, I get this error `"_video._capture initialization failed [1]: No incoming video signal detected"` on `hdmi_in.configure()`. Below is the complete set of code:

> from pynq import Overlay
> from pynq.lib.video import *
> base = Overlay('base.bit')
> hdmi_in = base.video.hdmi_in
> hdmi_in.configure()

I connected the HDMI input of PYNQ-Z1 from the HDMI output of my Laptop. I know for sure that my laptop's HDMI output is working fine. Has anyone also encountered this problem? How did you fix this? Is it fixed in the new API?

Hoping for your answer. Thanks.

shravan-red, 2019-10-19:
I have the exact same problem. Did you find a solution yet ?

timhae, 2020-03-04:
I resolved this issue by disabling the monitor and enabling it again using xrandr. I also had to set the resolution to 720p.

giunatale, 2020-03-04:
@timhae thanks for reporting out how you fixed this.
I am going to close this issue now. This is not a bug in PYNQ, it is something that happens with a few users and it is due to a misconfiguration and/or issues with the HDMI source. 
Please, for next time post this kind of questions on our support forum https://discuss.pynq.io

closed 2020-03-04 by giunatale