Xilinx/PYNQ #883: Zedboard Pynq v2.3 : Eth0 not found

shravan-red, 2019-08-06:
Because of my previous problem with overlay import where I couldn't download any bitstream file to my Zedboard while using the Pynq_Z1_v2.3 image on my SD card, I built a customized SD card pynq_v2.3 image for my Zedboard using PetaLinux 2018.2 with bionic.arm.2.3.img as base and avnet-digilent-zedboard-2018.2.bsp (both downloaded from Xilinx). The Zedboard booted from the SD card but I'm unable to connect to the local network through ethernet; on running ifconfig command only loopback address (127.0.0.1) is returned. There is no eth0. I can't use Jupyter Notebooks. Please suggest a solution

oceanraft, 2019-08-09:
may be devicetree? I'm using the Pynq_z2_v2.4 image on my zed, it's working for me.

## system-user.dtsi

/include/ "system-conf.dtsi"
/ {
	amba {

		audio0:audio-codec-ctrl@43c00000 {
			compatible = "xlnk,audio-codec-ctrl-1.0";
			reg = <0x43c00000 0x10000>;
		};

		ethernet@e000b000 {
			phy-handle = <&ethernet_phy>;
			ethernet_phy: ethernet-phy@0{
				reg = <0>;
			};
		};
	};
        usb_phy0: phy0 {
                compatible = "ulpi-phy";
                #phy-cells = <0>;
                reg = <0xe0002000 0x1000>;
                view-port = <0x170>;
                drv-vbus;
        };
};

&audio0 {
		compatible = "generic-uio";
};

&usb0 {
        usb-phy = <&usb_phy0>;
};

gsr-ee, 2019-08-14:
I've built the Pynq 2.3 and 2.4 for the Zedboard too. You can download from [here](https://1drv.ms/u/s!AtCaElnui86QgoIsYe0DQ0oOi5Wnqg?e=UJqvVp) and try. It ran without a problem on my Zedboard.

shravan-red, 2019-08-14:
Thank you very much, Gabriel. That solved my problem.

shravan-red, 2019-08-16:
> I built the Pynq 2.3 and 2.4 for the Zedboard too. You can download from [here](https://1drv.ms/u/s!AtCaElnui86QgoIsYe0DQ0oOi5Wnqg?e=UJqvVp) and try. It ran without a problem on my Zedboard.

shravan-red, 2019-08-16:
> may be devicetree? I'm using the Pynq_z2_v2.4 image on my zed, it's working for me.
>
> ## system-user.dtsi
> /include/ "system-conf.dtsi"
> / {
> amba {
>
> ```
> 	audio0:audio-codec-ctrl@43c00000 {
> 		compatible [...]

I made ethernet work on Zedboard by running the command **_sudo ifconfig eth0 172.30.1.90 netmask 255.255.255.0 up_** . In fact I have to run the same command every time the device is powered on. It is a minor inconvenience which could be avoided. Do you have a clue as to what might be the cause ?

m-pp, 2019-09-03:
With the 2.4 version of the image, the ethernet issue is resolved. But, HDMI on zedboard still doesn't work.

giunatale, 2019-09-27:
I will close this issue for now. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase, and this does not look like one.

Thanks in advance for your collaboration.

closed 2019-09-27 by giunatale

rassB, 2021-04-11:
Hi am trying to build PYNQ 2.6 image, i made my own bitstreams tcl hwh as files as in this [tutorial](https://discuss.pynq.io/t/tutorial-creating-a-hardware-design-for-pynq/145), i added the correct .spec files and built Zedboard images, but i am having some issues making the image boot on my board, i also tried making the boot leds blink and it doesn't seem to work. 
am i missing a critical step ?  
am quite a newbie, sorry if it's a very bad question am asking and thanks.

gsr-ee, 2021-04-23:
What is the kind of issues you are facing in the building of the pynq image for the zedboard?

[X]
[X]Regards,
Gabriel Silva Rocha.

vj-27, 2023-04-12:
> I built the Pynq 2.3 and 2.4 for the Zedboard too. You can download from [here](https://1drv.ms/u/s!AtCaElnui86QgoIsYe0DQ0oOi5Wnqg?e=UJqvVp) and try. It ran without a problem on my Zedboard.

the link is expired. can you please provide an updated link?

gauravacad, 2023-08-21:
> > I built the Pynq 2.3 and 2.4 for the Zedboard too. You can download from [here](https://1drv.ms/u/s!AtCaElnui86QgoIsYe0DQ0oOi5Wnqg?e=UJqvVp) and try. It ran without a problem on my Zedboard.

Please provide the new link to download