Xilinx/PYNQ #903: PYNQ Audio Streaming

rafferino, 2019-08-20:
I want to create a real time speech to text on the PYNQ. I have found the PYNQ Audio module accessible using the Base Overlay, but there doesn't seem to be a built in way to access the audio recording as a stream. I've thought of buffering the output of the `.record(time)` function using multiple threads but it seems I would be dropping some audio time in doing this. Is there a built in way to achieve this real time audio stream using existing libraries or should I look towards messing with the Audio IP in Vivado?

wady101, 2019-08-20:
https://github.com/wady101/PYNQ_Z2-Audio 
@rafferino you could probably have a go at this. I am still working on formatting it, but the source code is all there.

giunatale, 2019-09-27:
Hi @rafferino, I see your question was already answered. If you are still not satisfied, I need to kindly ask you to redirect your question to our forum: https://discuss.pynq.io
That's the dedicated place to receive support from the community.
The issue tracker here is to report bugs in the codebase.

Closing this now, thanks in advance for your collaboration.

closed 2019-09-27 by giunatale