Xilinx/PYNQ #949: CMA buffers cause crash when being viewed in Jupyter

FredKellerman, 2019-09-25:
This was experienced on Ultra96 v2.4 PYNQ, not sure if other platforms also behave this way. Or if this is a known issue that has been addressed.

If Xlnx() CMA buffer array contents are read for cell output display in a Jupyter Notebook it crashes the iPython kernel.  This also seems to happen if the contents are viewed with print(cma_buf) as well.  Doing things like the following in a cell cause the issue:

cma_buf
print(cma_buf)
a = cma_buf
print(a)
n = [1, 2, 3]
cma_buf = n

The work around is to make a copy into a Numpy array before viewing.  Like: np.array(cma_buf)

Accessing individual cma_buf elements with index works without issue.

PeterOgden, 2019-09-26:
This is an issue with non-cacheable memory which has a number of limitations in user-space that a lot of the Ubuntu standard library don't meet. If possible use cacheable memory (allocated by passing `cacheable=1` to `cma_array`) and manually handle flushing/invalidation and these issues won't arise.

FredKellerman, 2019-09-26:
Hi thank you for your reply.  Is cacheable memory suitable for DMA usage?

PeterOgden, 2019-09-26:
Yes - our `pynq.lib.DMA` driver will handle the flush/invalidate automatically or if the cma array has `.flush()` and `.invalidate()` functions you can use if you are using different IP.

closed 2019-09-26 by FredKellerman