Xilinx/PYNQ #964: DMA irq ports cause python empty string dict lookup

Recognition2, 2019-10-22:
**For general questions, please use [the PYNQ discussion forum](https://discuss.
pynq.io/). We actively monitor the forum and try to answer every question 
there.**

Before submitting a pull request, you can also raise an issue that explains the 
problem the pull request is trying to resolve. Feel free to close the issue once
 the pull request has been merged.

If you still want to raise an issue here, please give us as much detail as 
possible to the issue you are seeing - we've listed some helpful fields below.

* PYNQ version (e.g. v2.5): v2.5
* Board name (e.g. Pynq-Z1): Pynq-Z2
* Pull request type (bug fix / new feature): -
* Pull request number: -
* Description: Bug in IRQ usage.
Given a simple FIFO throughput setup, like this:

[image]

Given this setup, including the connection of the `s2mm_introut` and `mm2s_introut` pins to the `IRQ_F2P` using a `Concat` block, the pynq library will refuse to transmit any data over the DMA channels with the following error:
```
KeyError                                  Traceback (most recent call last)
<ipython-input-45-b53b94401f56> in <module>()
      1 import pynq.lib.dma
      2 
----> 3 dma_send = ol.axi_dma_ps_to_pl
      4 dma_recv = ol.axi_dma_pl_to_ps

/usr/local/lib/python3.6/dist-packages/pynq/overlay.py in __getattr__(self, key)
    335         """
    336         if self.is_loaded():
--> 337             return getattr(self._ip_map, key)
    338         else:
    339             raise RuntimeError("Overlay not currently loaded")

/usr/local/lib/python3.6/dist-packages/pynq/overlay.py in __getattr__(self, key)
    735         elif key in self._description['ip']:
    736             ipdescription = self._description['ip'][key]
--> 737             driver = ipdescription['driver'](ipdescription)
    738             setattr(self, key, driver)
    739             return driver

/usr/local/lib/python3.6/dist-packages/pynq/lib/dma.py in __init__(self, description, *args, **kwargs)
    188                                'has been deprecated and moved to '
    189                                'pynq.lib.deprecated')
--> 190         super().__init__(description=description)
    191 
    192         if 'parameters' in description and \

/usr/local/lib/python3.6/dist-packages/pynq/overlay.py in __init__(self, description)
    600             self._gpio = {}
    601         for interrupt, details in self._interrupts.items():
--> 602             setattr(self, interrupt, Interrupt(details['fullpath']))
    603         for gpio, entry in self._gpio.items():
    604             gpio_number = GPIO.get_gpio_pin(entry['index'])

/usr/local/lib/python3.6/dist-packages/pynq/interrupt.py in __init__(self, pinname)
     96         self.number = PL.interrupt_pins[pinname]['index']
     97         self.parent = weakref.ref(
---> 98             _InterruptController.get_controller(parentname))
     99         self.event = asyncio.Event()
    100         self.waiting = False

/usr/local/lib/python3.6/dist-packages/pynq/interrupt.py in get_controller(name)
    157             if con.name == name:
    158                 return con
--> 159         ret = _InterruptController(name)
    160         _InterruptController._controllers.append(ret)
    161         return ret

/usr/local/lib/python3.6/dist-packages/pynq/interrupt.py in __init__(self, name)
    175         """
    176         self.name = name
--> 177         self.mmio = MMIO(PL.ip_dict[name]['phys_addr'], 32)
    178         self.wait_handles = [[] for _ in range(32)]
    179         self.event_number = 0

KeyError: ''
```
This is caused by the interrupt pin being created in the static method `_Interrupt.get_controller`, more specifically, the `parent` of the interrupt being an empty string.
This results in a lookup in `ip_dict` with an empty string as key, which fails.
If I remove the interrupts from the block diagram (and from the ZYNQ7 PS) then this code does run, but does not do throughput. Instead, it always outputs `dec0dee3`, and I have no clue why.

This is using Vivado 2019.1, pynq 2.5, the PYNQ Z2 board, linux 4.14-xilinx-v2018.3.

iapapistas, 2019-12-05:
Indeed I am facing the same error. Initially I found that adding an AXI interrupt controller between the PS7 and the AXI peripheral will solve this particular problem, however then a new error appears: 
> ValueError: Could not find UIO device for interrupt pin for IRQ number 0

Did you eventually find any workaround for the initial issue?

iq-embedinn, 2020-02-04:
please check weather linux uio driver  is setup and installed properly?
$ls /dev/uio0
if its not present then you need  to update the devicetree.
or linux kernel module for uio  driver etc.

twaclaw, 2020-03-18:
Have a look at this entry:

https://forums.xilinx.com/t5/Embedded-Linux/A-key-error-when-I-am-trying-to-access-a-DMA-IP-on-PYNQ-Z1-board/td-p/847817

mariodruiz, 2021-06-02:
For more than one interrupt you need to use an AXI Interrupt Controller 

https://discuss.pynq.io/t/interrupt-error-on-using-two-vdma/2700/4