Xilinx/PYNQ #965: Connecting Pynq Z2 to wifi

Katherine639, 2019-10-23:
Hi,

So I am having trouble connecting the board to wifi. I have tried the ethernet cable from laptop to board and using wifi, i tried bridging connection but in the end I end up with unidentified network and then no internet.

I have also tried connecting the board to router and then doing the http:// but it is still not working.

I am out of ideas as I have already looked up the tutorials on how to connect the board to wifi and such. Any help or clue on how to get this board started so i can used Jupyter notebook?

MarinMikael, 2019-10-23:
Hello,

I do not use WIFI on Pynq, but here are two ways to access the board through Ethernet that I commonly use:

1) Ethernet: you need a router with DHCP, if everything works, the board's jupyter should be accessible on http://zynq:9090 after one or two minutes

2) Access the terminal though UART (baud: 115200). 
If you use windows, check the device manager, and see which COM ports appears when you plug the board. Then use any tool (putty, mobaxterm, or other).
From here, you will have full access to the operating system to configure the ip addresses and wifi configuration.

In addition, make sure you flash the latest Pynq image on the microsd card, to use the latest features and make sure it corresponds to the documentation. If 2) show errors on the terminal, making a fresh microsd card will most likely solve the problem.

Mikaël

MarinMikael, 2019-10-23:
Oh, and I forgot. The wifi part will most likely depend on your wifi dongle and just follow the regular process. It does not depend on the pynq, and will most likely require an access through ssh or uart to configure it. 

Mikael

giunatale, 2019-10-23:
Hi,

The best way to provite internet access to the board is to use a WiFi dongle directly connected through USB to the Z2, as it is in general pretty straightforward. 
Ethernet from laptop to board is usually intended for accessing the board itself, as documented [here](https://pynq.readthedocs.io/en/v2.5/getting_started/pynq_z2_setup.html#connect-to-a-computer). Having "unidentified network" if you [assign your computer a static ip](https://pynq.readthedocs.io/en/v2.5/appendix.html#assign-your-computer-a-static-ip) to connect to the board, it is normal and expected. Bridging the connection to internet from your laptop to the board through ethernet is possible but it is not a use case we explicitly support or provide instructions for.
It is also not clear whether you want to provide internet access to the board, or you just want to access the board from your laptop to use jupyter. If it is the latter, @MarinMikael already provided some excellent tips. You can find more detailed instructions on our [documentation page](https://pynq.readthedocs.io/en/v2.5/getting_started/pynq_z2_setup.html) dedicated to the Z2.

Regardless, since this is an issue that is not related to the code base, but rather a support/general question, you should move to https://discuss.pynq.io. I will have to close this now. For the future, please carefully follow the instructions on the [issue template](https://github.com/Xilinx/PYNQ/blob/master/.github/ISSUE_TEMPLATE/pynq-issue-template.md) that is presented when opening a new issue.

closed 2019-10-23 by giunatale