Xilinx/PYNQ #976: Tcl parsing issue Vivado 2019.2

cathalmccabe, 2019-12-06:
In Vivado 2019.2 IPI the default Tcl command that is run to assign IP addresses seems to have changed from `assign_bd_address` in previous versions to `assign_bd_address`. 
This command is auto run when connection automation is used to connect a salve IP. 
This command pattern is used in Tcl parsing, so breaks the generation of the ip_dict. 

Some info on issue raised in forum here:
https://discuss.pynq.io/t/overlay-loading-problem-ip-dict-empty/93/5

The commands are not direct replacements for each other and have a different format.  
Ref:https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_2/ug835-vivado-tcl-commands.pdf

create_bd_addr_seg range <arg> offset <arg> [quiet] [verbose]
 [<parent_addr_space>] [<slave_segment>] <name>
  
assign_bd_address [target_address_space <arg>] [boundary]
 [master_boundary] [external] dict <arg> [import_from_file <arg>]
 [export_to_file <arg>] [export_gui_to_file <arg>] [offset <arg>]
 [range <arg>]
 
2019.1 example:

create_bd_addr_seg -range 0x20000000 -offset 0x60000000 [get_bd_addr_spaces iop_pmodb/mb/Data] [get_bd_addr_segs ps7_0/S_AXI_GP0/GP0_M_AXI_GP0] SEG_ps7_0_GP0_M_AXI_GP0

2019.2 example:

assign_bd_address -offset 0x43C00000 -range 0x00010000 -target_address_space [get_bd_addr_spaces processing_system7_0/Data] [get_bd_addr_segs scalar_add/s_axi_AXILiteS/Reg] -force

maersdal, 2019-12-06:
Can confirm. This also happens with the base.tcl script and Vivado 2019.2 Using the makefile workflow.
2019.1:
`# Create address segments`
`
  create_bd_addr_seg -range 0x00010000 -offset 0x43C00000 [get_bd_addr_spaces ps7_0/Data] [get_bd_addr_segs audio_direct_0/S_AXI/S_AXI_reg] SEG_audio_direct_0_S_AXI_reg
  `

2019.2 : 
`# Create address segments`
`
  assign_bd_address -offset 0x43C00000 -range 0x00010000 -target_address_space [get_bd_addr_spaces ps7_0/Data] [get_bd_addr_segs audio_direct_0/S_AXI/S_AXI_reg] -force
  `

patocarr, 2019-12-08:
I've made a few modifications to tcl_parser in an attempt to handle the new formatting on 2019.2. It seems to be working for my [limited] use case.

[tcl_parser.txt](https://github.com/Xilinx/PYNQ/files/3937260/tcl_parser.txt)

giunatale, 2020-11-02:
I am closing this issue as we officially deprecated TCL with the 2.6 release

closed 2020-11-02 by giunatale