Xilinx/PYNQ #983: Updating Pynq loses base.bit

ralphrmartin, 2020-01-08:
I tried updating Pynq using 
`sudo pip3 install --upgrade git+https://github.com/Xilinx/PYNQ.git`
as per the readme, and as a result I no longer have base.bit (and the hwh file I suppose, and I presume the same goes for the  logictools overlay too.). 

This is not very friendly behaviour! I understand they may not be included because of size etc, but it would be more user-friendly to tell git to preserve these files when doing an update.

giunatale, 2020-01-09:
Hi @ralphrmartin,
We apologize for your unfortunate experience. The readme on the github repo is not updated, but as a note, from PYNQ v2.5 we removed the bitstreams from the repo, so installing PYNQ from the repo itself won't work!
As part of the [v2.5 release](https://github.com/Xilinx/PYNQ/releases/tag/v2.5) we distributed the source distribution [tarball](https://github.com/Xilinx/PYNQ/releases/download/v2.5/pynq-2.5.tar.gz) that you'll have to download and install via
`sudo pip3 install --upgrade pynq-2.5.tar.gz`

giunatale, 2020-01-09:
I have reported this internally, we will fix the out-of-sync README, thanks for letting us know about this!
I will close the issue now

closed 2020-01-09 by giunatale