Xilinx/PYNQ #989: 18.04 host support still broken

powderluv, 2020-01-22:
This is a follow up to pull request #974 

We removed the check for 18.04 packages at the last minute when we merged #974 so if you build on a native host you will get an error. 

If you install gcc-multilib as petalinux suggests and try to build you will get this error:

foo@bar:~/github/PYNQ/sdbuild$ time make BOARDDIR=/home/foo/github/Ultra96-PYNQ BOARDS=Ultra96
/opt/qemu/bin/qemu-aarch64-static -version | fgrep 4.0.0
qemu-aarch64 version 4.0.0                                     
vivado -version | fgrep 2019.1   
Vivado v2019.1 (64-bit)                                             
xsdk -version | fgrep 2019.1    
****** SDK v2019.1 (64-bit)                                                                                                                
which petalinux-config                                                                                                        
/home/foo/peta/tools/common/petalinux/bin/petalinux-config
which arm-linux-gnueabihf-gcc                                           
Makefile:342: recipe for target 'checkenv' failed
make: *** [checkenv] Error 1                                      
                                                                        
real    0m2.204s                                                                            
user    0m2.036s                                                      
sys     0m0.242s                    

If you install gcc-arm-linux-gnueabihf you will uninstall gcc-multilib above and fail:

You will get this error:

libsdl1.2-dev
rsync
python3-pip
gcc-multilib
Error: Package not found - gcc-multilib
Run setup_host.sh
Makefile:342: recipe for target 'checkenv' failed
make: *** [checkenv] Error 1

It is a trivial patch and I'm happy to resubmit it but need your guidance @yunqu @giunatale since I don't want to submit a PR that has no chance of getting to master.

yunqu, 2020-01-24:
Hi, I have run the following:

```
which arm-linux-gnueabihf-gcc
```
And it gives me:

`/sdbuild/Xilinx/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin/arm-linux-gnueabihf-gcc`

So I guess the reason why you think the current branch is broken is that you have not installed Xilinx SDK, or forget to source SDK settings?

powderluv, 2020-01-24:
I have sourced it and all tools including xsdk are in the path. Please see output below 

/opt/qemu/bin/qemu-aarch64-static -version | fgrep 4.0.0
qemu-aarch64 version 4.0.0
vivado -version | fgrep 2019.1
Vivado v2019.1 (64-bit)
xsdk -version | fgrep 2019.1
****** SDK v2019.1 (64-bit)
which petalinux-config

powderluv, 2020-01-24:
Wait /sdbuild only exists in your vagrant setup? It wouldn't if you ran it natively with vagrant I assume. I'm trying to run on a native Ubuntu machine so I don't have to use vagrant and it's CPU emulator

yunqu, 2020-01-24:
It shouldn't matter. /sdbuild is only a folder created during the vagrant setup. You can use another name on your native machine. 

I tried `which arm-linux-gnueabihf-gcc` on one of our build machines (not the VM on my desktop), and it showed:

`<installation_path>/SDK/2019.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin/arm-linux-gnueabihf-gcc`

It looks consistent over multiple machines. I was wondering your problem could be more specific to your environment. Can we move further discussion of this topic over to discuss.pynq.io?

powderluv, 2020-01-24:
You are right. You need to source the settings64.sh also to get it in the path. Closing.

closed 2020-01-24 by powderluv