Xilinx/PYNQ #993: NFS does not for Ultra96 V1, PYNQ V2.5

yahoo2016, 2020-01-25:
I flashed latest PYNQ and ran:
`apt update`
`apt install nfs-kernel-server`
I got:
pynq systemd[1]:nfs-server.service: Dependency failed for NFS server and service
pynq systemd[1]:nfs-server.service: Job nfs-server-service/start failed with result "dependency" 
I tried `systemctl restart nfs-kernel-server` and got the same errors.
I tried:
`apt purge nfs-kernel-server`
`apt upgrade`
`apt install nfs-kernel-server`
and got the same errors.

giunatale, 2020-01-27:
Hi @yahoo2016, do you have a way to test on another PYNQ board? In case this problem is specific to the Ultra96 V1, I'd recommend you open an issue on the official PYNQ-Ultra96 repo:
https://github.com/Avnet/Ultra96-PYNQ

yahoo2016, 2020-01-29:
Hi @giunatale,
I don't have other type of PYNQ board.  I wish other developers can prove or disapprove NFS server work or not on PYNQ boards.
I posted the issue at Ultra96-PYNQ repo.
Thanks

giunatale, 2020-01-29:
I would suggest that for now you redirect the question to our support forum https://discuss.pynq.io. As of now we cannot confirm this is a bug so I'll have to close the issue.

Please, open a thread on the forum and post a complete log of the `nfs-kernel-server` service. We can start from there.

closed 2020-01-29 by giunatale