Xilinx/PYNQ #994: we get the non intenet and cant connect error on getting the version of the code from one computer to another

vishwas1234567, 2020-01-26:
**For general questions, please use [the PYNQ discussion forum](https://discuss.
pynq.io/). We actively monitor the forum and try to answer every question 
there.**

Before submitting a pull request, you can also raise an issue that explains the 
problem the pull request is trying to resolve. Feel free to close the issue once
 the pull request has been merged.

If you still want to raise an issue here, please give us as much detail as 
possible to the issue you are seeing - we've listed some helpful fields below.

* PYNQ version (e.g. v2.5):
* Board name (e.g. Pynq-Z1):
* Pull request type (bug fix / new feature):
* Pull request number:
* Description:

vishwas1234567, 2020-01-26:
[image]

vishwas1234567, 2020-01-26:
[image]

is this ok after the installation.

vishwas1234567, 2020-01-26:
not that complete thing but thus 
[image]

vishwas1234567, 2020-01-26:
this all happened when we did this process until then the boar was working fine.
[image]

vishwas1234567, 2020-01-26:
https://docs.google.com/document/d/1JnOTYO-54EFyPGTecnlRaJsA72EqhBG_f69VFIz73nM/edit?usp=sharing

the thing we found working can be found here.

giunatale, 2020-01-27:
Hi @vishwas1234567, please adhere to the issue template for a next time.
Secondly, as this seems more of a support question, I will close this now and ask you to redirect your query to https://discuss.pynq.io/

Another request, try to comment the issue, using only screenshots is not going to make things clear.

Many thanks!

closed 2020-01-27 by giunatale