Xilinx/PYNQ #998: The pynq module cant run the setup.py for the working

vishwas1234567, 2020-01-30:
[image]

[image]

[image]

giunatale, 2020-01-30:
Hi @vishwas1234567, as I asked you already in https://github.com/Xilinx/PYNQ/issues/994, sending just screenshots is not gonna be helpful. A clear explanation of the issue in text is **strictly required**. Also, we ask users to adhere to the issue template that is displayed when you open a new issue.

To reply your question: pip-installing from the repo is going to be deprecated and is not gonna work starting from v2.5. This is not yet documented on readthedocs or in the readme, we are already aware of it but we plan on fixing it soon.

You will have to update pynq using the [tarball](https://github.com/Xilinx/PYNQ/releases/download/v2.5/pynq-2.5.tar.gz) released as part of [v2.5](https://github.com/Xilinx/PYNQ/releases/tag/v2.5).
Download the tarball and then do
```bash
sudo pip3 install --upgrade --upgrade-strategy only-if-needed pynq-2.5.tar.gz
```
Don't forget the `--upgrade-strategy only-if-needed`, that's also very important.

Closing the issue now.

closed 2020-01-30 by giunatale

vishwas1234567, 2020-01-30:
sir can you send the video link.
[image]

vishwas1234567, 2020-01-30:
we still get the same error

giunatale, 2020-01-30:
Can you please post on https://discuss.pynq.io? Issues are not meant for technical support.

The error is self-explanatory: no video signal is detected, make sure your set-up is correct and a video source is actually going through HDMI in.
Again though, please post your question on the support forum.
When you do that, explain **in detail** your setup (what you connected to the board, how you configured everything, PYNQ version, SD card image version etc...), what you are trying to achieve, and what did you do to try to fix it.

Please avoid just sending screenshots, they are not helpful.

vishwas1234567, 2020-01-30:
Sir the setup was done the image file also came on the monitor bit after
sometime it didn't work and showed two error after running each time

Lookuperror and valueerror