Xilinx/PYNQ #999: Problem about connecting ethernet when setup by using static IP

VulpesLagopus, 2020-02-01:
In my case, I connect directly to my computer, and I assign my computer a static IP address like 192.168.2.1, and connect the board to my pc's ethernet port. Of course, the microusb port is also connected to the pc. 

But when I turn on the PYNQ-Z2, the RED LED comes immediately and the Yellow/Green Done LED does also. But the LED on the ethernet port both on the board and my pc is not on. And then the Done LED starts to flash slowly and there is no two BLUE LEDs and four Yellow/Green LEDs.

It seems the ethernet does not work totally. Sometimes I can setup smoothly without this situation.  But sometimes I face this problem. Does anyone face the same situation like me? Please!

VulpesLagopus, 2020-02-01:
PS. I can find the COM in device manager so I think the microusb is connected smoothly.

giunatale, 2020-02-03:
Please follow the issue template when posting issues.

As this is a support question, I am asking you to redirect this to our support forum https://discuss.pynq.io.

Closing

closed 2020-02-03 by giunatale