Xilinx/PYNQ #1017: CIFS module for PYNQ Z1

aadeocs, 2020-03-09:
I need to access a Windows shared folder from PYNQ Z1.

For Ubuntu Desktop18.04 on Intel, this sequence works:
$ sudo apt-get install nfs-common
$ sudo apt install cifs-utils
Then this command (after creating /mnt/drivem):
$ mount //192.168.1.100/drivem /mnt/drivem -o user=beagles,rw,gid=1000,uid=1000 

On a fresh PYNQ 2.5 that sequence yields:
> mount error: cifs filesystem not supported by the system
> mount error(19): No such device
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) 

Apparently the CIFS module is missing:
$ modprobe cifs
Yields:
> modprobe: FATAL: Module cifs not found in directory /lib/modules/4.19.0-xilinx-v2019.1

Any information on getting CIFS installed, or when it may be installed in PYNQ, would be appreciated.

Thanks,
Dave

giunatale, 2020-03-09:
Please follow the issue template directives for a future issue. For this kind of questions, post on our [support forum](https://discuss.pynq.io/c/support). 
The issue tracker here is meant to reeport bugs in the codebase.

Thanks for your collaboration.
Closing.

closed 2020-03-09 by giunatale

aadeocs, 2020-03-09:
Why closed?

giunatale, 2020-03-09:
@aadeocs see my reply. Please use https://discuss.pynq.io for support questions that do not report bugs in the code, as in this case. This is clearly written in the issue template that is presented when opening a new issue.

Thanks again for your collaboration.

aadeocs, 2020-03-09:
My first post was here:
> https://discuss.pynq.io/t/access-windows-share-from-pynq-z1/977,
   Where I was redirected to a TBD linux forum
Then I posted here:
>
https://forum.digilentinc.com/topic/19704-access-windows-share-from-pynq-z1/#comment-54197
   It is clear that you need to get or build the `cifs` module.
   Maybe you can find something related to kernel modules on PYNQ webpage.
No response from Xilinx from three days ago:
>
https://forums.xilinx.com/t5/Embedded-Linux/PYNQ-Z1-Access-Windows-shared-folder/m-p/1082551#M40790

Not clear to me this is a bug, it may be a module that was not included in
the release?
> modprobe: FATAL: Module cifs not found in directory
/lib/modules/4.19.0-xilinx-v2019.1

Sorry if I posted in the wrong location, not sure where else to post.
Aside from a linux rebuild, there doesn't appear to be a way to implement
this common feature.

Thanks,
Dave

giunatale, 2020-03-09:
You are right indeed, you will have to rebuild the kernel and re-create the SDCard image to add the feature you request. You can do it yourself and follow the various guides we provide on how to rebuild our PYNQ SDCard image:

https://github.com/Xilinx/PYNQ/blob/master/sdbuild/README.md
https://pynq.readthedocs.io/en/v2.5.1/pynq_sd_card.html

aadeocs, 2020-03-09:
guinatale - thank you very much - Dave