Xilinx/PYNQ #1033: Looking to update Overlay Design Methodology

rogerpease, 2020-05-17:
I'm an experienced hardware and software designer  (mostly ASIC and some FPGA). I have a Tul-2 board I use for hobby purposes. I'd like to update the sections of https://pynq.readthedocs.io/en/v2.5.1/overlay_design_methodology.html for experienced Verilog programmers like myself. 

I learned about the AXI interface through https://www.xilinx.com/video/hardware/creating-an-axi-peripheral-in-vivado.html and would like to merge some updates into either the existing or a new one of https://pynq.readthedocs.io/en/v2.5.1/overlay_design_methodology/overlay_tutorial.html (maybe making verilog_overlay_tutorial.html ?)

Any toes I am stepping on/suggestions/comments?

rogerpease, 2020-06-23:
I wrote this and put it in a branch for review. 

One thing ... from the Current overlay page it looks like there is a way to use these pragmas (from the original page) 

void add(int a, int b, int& c) {
#pragma HLS INTERFACE ap_ctrl_none port=return
#pragma HLS INTERFACE s_axilite port=a
#pragma HLS INTERFACE s_axilite port=b
#pragma HLS INTERFACE s_axilite port=c

    c = a + b;
}

to 'short circuit' AXI IP creation and have the AXI tool figure out that a=register offset 0x0, b=register offset 0x04 and c= register offset 0x08 (so you don't need to hand-edit the AXI peripheral the way I did). Is that so?? If so does that process exist for Verilog? Can you point me to the equivalent?

MarinMikael, 2020-06-30:
Hello,
I never used custom verilog IP with Pynq, however, I do use HLS quite a lot with Pynq and I ca provide some answers.

Those pragma in HLS because HLS use transactional interfaces (TLM) rather than signal and register interfaces (RTL). I am not aware of any TLM feature in Verilog. But I only use VHDL and HLS, not Verilog..

To identify the list of registers, Pynq relies on your Vivado project's HDF file. 
Hence, a Verilog IP packaged with Vivado into a proper AXI4-lite IP should properly show-up in the memory mapping on Vivado and in the HDF file, and finally be properly identified by Pynq.

Please refer to [pg1118](https://china.xilinx.com/support/documentation/sw_manuals_j/xilinx2020_1/ug1118-vivado-creating-packaging-custom-ip.pdf) for a detailed documentation on custom IP packaging on Vivado.