Xilinx/PYNQ #1044: doc: an easier way to write img to SD card

jbattat, 2020-06-05:
This page:
[https://pynq.readthedocs.io/en/v1.3/17_appendix.html#writing-the-sd-card-image](https://pynq.readthedocs.io/en/v1.3/17_appendix.html#writing-the-sd-card-image)
outlines a process on a Mac to write a disk image file to an SD card using command line tools. 

Another, potentially useful and very straightforward tool is Apple Pi Baker (used widely in the Raspberry Pi community).

Download/install Apple Pi Baker v2 (v1 also works on macOS Catalina):
[https://www.tweaking4all.com/hardware/raspberry-pi/applepi-baker-v2/#DownloadApplePiBaker](https://www.tweaking4all.com/hardware/raspberry-pi/applepi-baker-v2/#DownloadApplePiBaker)

Insert SD card, click "Select Disk(s)" and choose the SD card.  Then click "Restore" and select the .img file (e.g. `pynq-z2_v2.5.img`). Click "Open".  Wait a few minutes. Done.

VicerExciser, 2020-10-21:
I think [Balena Etcher](https://www.balena.io/etcher/) should be recommended over Apple Pi Baker.

giunatale, 2020-11-02:
Hi Guys, thanks for this suggestions, but the aim of the documentation there is to use what's already available off-the-shelf. Balena Etcher is indeed a tool we also use internally, but it requires the tool to be installed, which is not a bad thing but it is an additional step.

I would encourage though one of you to write a PR where you provide this proposed changes. This is an open source project and we welcome any contribution. We will review and consider merging what you propose if we deem it useful.

Thanks, closing

closed 2020-11-02 by giunatale