Xilinx/PYNQ #1174: VItis HLS error

Wo-Ki, 2020-12-20:
* PYNQ version (e.g. v2.5): v2.6
* Board name (e.g. Pynq-Z1): pynq-z2
* Pull request type (bug fix / new feature): bug fix
* Ubuntu 19.04
* Vivado 2020 02(vitis hls instead of vivado hls)
* Description:
When I run**./build_ip.sh**, some errors like:
`ERROR: [HLS 200-644] The 'set_part -tool' hidden command is not supported.
command 'set_part' returned error code
    while executing
"set_part {xc7z020clg400-1} -tool vivado"
    (file "pixel_unpack/script.tcl" line 6)
    invoked from within
"source pixel_unpack/script.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
`
like:
`command 'ap_source' returned error code
    while executing
"source trace_cntrl_32/script.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
`
And, there are 23 IPs generated in result(27 totally).I don't know how to solve it, thank for your help.

mariodruiz, 2020-12-21:
Hi @Wo-Ki,
The latest PYNQ version (v2.6) has been verified using Vivado 2020.1
I suggest to install Vivado 2020.1, we are not supporting 2020.2 at this point.

Mario

Wo-Ki, 2020-12-21:
I installed vivado 2019.2，and it worked.Thank you very much.

closed 2020-12-23 by Wo-Ki