Xilinx/PYNQ #1178: Some errors in the document.

liuwilliamBUPT, 2021-01-15:
* PYNQ version (e.g. v2.5): 2.5
* Board name (e.g. Pynq-Z1): Pynq-Z2
* Description:  I try to follow the example, however, there may be two mistakes in that. 
* 
When the following code ran, I got a KeyError in Jupyter which is `KeyError: 'gpio_leds'`.

I find that the above document mentions that we should use `[led|switch|button|rgbleds]_gpio`, and I do that. 

Then I find it's ok without errors. So I think these two lines may have some errors. 

I think`gpio_leds` should be modified to `leds_gpio` and `gpio_switches` to `switches_gpio`.

* The code
https://github.com/Xilinx/PYNQ/commit/82da59069aa56f62dbc7dda484b6ff2f805b60cc#diff-9b7aa25cb5b22e220f72782035e943209a204ecac4db6ffab0cc0a588bc920e4R67
https://github.com/Xilinx/PYNQ/commit/82da59069aa56f62dbc7dda484b6ff2f805b60cc#diff-9b7aa25cb5b22e220f72782035e943209a204ecac4db6ffab0cc0a588bc920e4R67

* The document
https://github.com/Xilinx/PYNQ/commit/82da59069aa56f62dbc7dda484b6ff2f805b60cc#diff-9b7aa25cb5b22e220f72782035e943209a204ecac4db6ffab0cc0a588bc920e4R67

tomvocas6, 2021-06-25:
Hallo ,  thank you for commet. Wery useful for me.  For my board  pynq-z2,  was  btns  working instead of buttons.

liuwilliamBUPT, 2021-06-25:
> Hallo , thank you for commet. Wery useful for me. For my board pynq-z2, was btns working instead of buttons.

You're welcome. But I think this repo may have stopped maintenance so long, lol.

PR #1241 (2021-08-30): Fixes in AxiGPIO documentation and code
This fixes #1178