Xilinx/PYNQ #1285: Jupyter package requires a more recent NodeJS installation

datMaffin, 2021-10-14:
* PYNQ version:
  - branch origin/image_v2.6.2
* Board name:
  - zedboard (should be irrelevant)
* Description:
  
  Running the image creation resulted in an npm error, which turns out to be an error because a dependency requires a NodeJS version that is greater than the available version.
  
  To fix this issue, I kinda put a bandage around the problem by changing the following, which is going to install a more recent `nodejs`:
  ```diff
  diff --git a/sdbuild/packages/jupyter/qemu.sh b/sdbuild/packages/jupyter/qemu.sh
  index 28e7ffd6..b465f6ab 100755
  --- a/sdbuild/packages/jupyter/qemu.sh
  +++ b/sdbuild/packages/jupyter/qemu.sh
  @@ -13,6 +13,12 @@ else
          export NODE_OPTIONS=--max-old-space-size=4096
   fi
   
  +# Install a recent nodejs for jupyter
  +echo deb https://deb.nodesource.com/node_14.x bionic main > /etc/apt/sources.list.d/nodesource.list
  +apt-get update
  +apt-get install -y --allow-unauthenticated nodejs
  +
  +
   jupyter notebook --generate-config --allow-root
   
   cat - >> /root/.jupyter/jupyter_notebook_config.py <<EOT
  ```

datMaffin, 2021-10-15:
Nice to see! 

However, I am sorry that I missed the changes in the 2.7 version branch.

schelleg, 2021-10-19:
Thank you for finding it and the nice report ... we too had to upgrade, release should be out soon...

closed 2021-10-19 by schelleg