Xilinx/PYNQ #1286: Trying to build the rootfs from scratch fails because ct-ng toolchain component `isl-0.20` tarball fails to download.

datMaffin, 2021-10-15:
* PYNQ version:
  - branch origin/image_v2.6.2
* Board name:
  - zedboard (should be irrelevant)
* Description:
  Image creation fails when not using a base rootfs image, i.e., running `make` in `sdbuild` without specifying `PREBUILT`.
  ```
  Now configured for "arm-unknown-linux-gnueabihf"
  make[1]: Leaving directory '/home/user/Documents/PYNQ/sdbuild/build/gcc-mb'
  make[1]: Entering directory '/home/user/Documents/PYNQ/sdbuild/build/gcc-mb'
  [INFO ]  Performing some trivial sanity checks
  [WARN ]  Number of open files 1024 may not be sufficient to build the toolchain; increasing to 2048
  [INFO ]  Build started 20211015.120804
  [INFO ]  Building environment variables
  [WARN ]  Directory '/home/dostalmn/src' does not exist.
  [WARN ]  Will not save downloaded tarballs to local storage.
  [EXTRA]  Preparing working directories
  [EXTRA]  Installing user-supplied crosstool-NG configuration
  [EXTRA]  =================================================================
  [EXTRA]  Dumping internal crosstool-NG configuration
  [EXTRA]    Building a toolchain for:
  [EXTRA]      build  = x86_64-pc-linux-gnu
  [EXTRA]      host   = x86_64-pc-linux-gnu
  [EXTRA]      target = arm-unknown-linux-gnueabihf
  [EXTRA]  Dumping internal crosstool-NG configuration: done in 0.06s (at 00:01)
  [INFO ]  =================================================================
  [INFO ]  Retrieving needed toolchain components' tarballs
  [EXTRA]    Retrieving 'automake-1.16.1'
  [EXTRA]    Verifying SHA512 checksum for 'automake-1.16.1.tar.xz'
  [EXTRA]    Retrieving 'linux-4.20.8'
  [EXTRA]    Verifying SHA512 checksum for 'linux-4.20.8.tar.xz'
  [EXTRA]    Retrieving 'zlib-1.2.11'
  [EXTRA]    Verifying SHA512 checksum for 'zlib-1.2.11.tar.xz'
  [EXTRA]    Retrieving 'gmp-6.1.2'
  [EXTRA]    Verifying SHA512 checksum for 'gmp-6.1.2.tar.xz'
  [EXTRA]    Retrieving 'mpfr-4.0.2'
  [EXTRA]    Verifying SHA512 checksum for 'mpfr-4.0.2.tar.xz'
  [EXTRA]    Retrieving 'isl-0.20'
  [ERROR]    isl: download failed
  ...
  ```

  See also: https://github.com/crosstool-ng/crosstool-ng/issues/1609

schelleg, 2021-10-19:
Thanks for the heads up... its an important FYI and we'll try to do more CI between releases at least to give everyone the warning that something disappeared and then we can provide alternate solutions

closed 2021-10-19 by schelleg

keringar, 2021-11-12:
I needed to build a sysroot so I couldn't use the base file system. I was able to work around this by modifying the sdbuild/packages/gcc-mb/Makefile to include a sed command that changed the isl and expat mirrors as outlined here: https://github.com/crosstool-ng/crosstool-ng/issues/1625