Xilinx/PYNQ #1319: Update interrupt document with the fix for every second interrupt call failing

jancumps, 2021-11-19:
Forum post [https://discuss.pynq.io/t/zynq-ps7-interrupts/198](https://discuss.pynq.io/t/zynq-ps7-interrupts/198) has a solution for an issue with the interrupt design.
The interrupt works the first time a Python function waits for it. The second time, a bogusinterrupt is picked up.
Community member datMattin posted a fix that I tested: [https://discuss.pynq.io/t/zynq-ps7-interrupts/198/10](https://discuss.pynq.io/t/zynq-ps7-interrupts/198/10)

I ammended the original documentation with the step to fix this, and an image of the Vivado property page. I submitted pull request #1318

jancumps, 2021-11-22:
Pull request merged. Thank you.

closed 2021-11-22 by jancumps

schelleg, 2021-11-22:
Thank you for the contribution!