Xilinx/PYNQ #1322: Pynq 2.7 for ZCU208

sirjansel0t, 2021-11-24:
According to the release notes for pynq 2.7 it supports rf data converters and clocks for the gen3.  Does pynq 2.7 support the zcu208 or any other gen 3 eval board?

jramsey123, 2022-03-12:
Has there been any progress made on getting PYNQ running on the ZCU208?

plysaght, 2022-03-12:
PYNQ support for the ZCU208 is not currently plan of record but it is being reviewed. If would be helpful to gather any use cases that would you can offer to make the case for doing th port. 
FYI there is a 3rd-party port of PYNQ for the ZCU216 here: https://github.com/sarafs1926/ZCU216-PYNQ

mariodruiz, 2022-10-17:
Starting from PYNQ v3.0, the ZCU208 is supported in the RFSoC-PYNQ repository. https://github.com/Xilinx/RFSoC-PYNQ/tree/master/boards/ZCU208

closed 2022-10-17 by mariodruiz