Xilinx/PYNQ #1325: Overlay tutorial

j3parera, 2021-12-30:
Hi,
I'm trying the Overlay tutorial as stated in the documentation using PYNQ-Z2 and Vivado 2020.2, Vitis_HLS 2020.2.
I've managed to make the first part work (scalar_add). But I can't make the second part (multiply and transfer using DMA) work. As far as I can see there are some issues with the DMA: no TLAST defined, buffer unaligment...
Could you, please, upgrade the tutorial?
Thanks in advance,
José Parera

EmbeddedTec, 2021-12-31:
Hi, the overlay tutorial is located in another git repo: https://github.com/schelleg/overlay_tutorial
I created a pull request there, that solves the problem.

The overlay_tutorial was formerly managed by PeterOgdon. Now Graham Schelle (@schelleg) manages the turorial.
The link in the readthedocs tutorial should be linked to his repo, so it can be found.

Best regards

EmbeddedTec, 2021-12-31:
I created a pull request regarding the docs page: https://github.com/Xilinx/PYNQ/pull/1326

j3parera, 2022-01-01:
Ok. Thanks a lot. I'll try it.

j3parera, 2022-01-02:
Hi,
I've been trying to build the tutorial code for the PYNQ-Z2 board. I've changed the board definition in the scripts. But the compilation of tutorial 1 ends with an error as a consequence of this critical warning:
`CRITICAL WARNING: [Project 1-486] Could not resolve non-primitive black box cell 'tutorial_1' instantiated as 'tutorial_1_i' [e:/overlay_tutorial/overlays/tutorial_1/tutorial_1.gen/sources_1/bd/tutorial_1/hdl/tutorial_1_wrapper.v:16]`
Is this due as not using a Digilent board?
Thanks in advance and best regards.

EmbeddedTec, 2022-01-02:
My Vitis HLS Build fails since yesterday, too:
`source run_ippack.tcl -notrace`
`bad lexical cast: source type value could not be interpreted as target`
`    while executing`
`"rdi::set_property core_revision 2201022152 {component component_1}"`
`    invoked from within`
`"set_property core_revision $Revision $core"`
`    (file "run_ippack.tcl" line 826)`
`INFO: [Common 17-206] Exiting Vivado at Sun Jan  2 21:52:44 2022...`
`ERROR: [IMPL 213-28] Failed to generate IP. `

The build still worked in 2021. It seems to be a year date bug:
https://support.xilinx.com/s/question/0D52E00006uxy49SAA/vivado-fails-to-export-ips-with-the-error-message-bad-lexical-cast-source-type-value-could-not-be-interpreted-as-target?language=en_US

Does setting the system date to 2021 fix the problem for you?

j3parera, 2022-01-03:
What an ugly workaround! But it worked. :)
Thank you so much.

skalade, 2024-05-01:
Closing due to lack of activity. If this is still an issue re-open it as a [discuss.pynq.io](https://discuss.pynq.io/) post.

closed 2024-05-01 by skalade