Xilinx/PYNQ #1328: errors when generating board design for  ZCU104 using base.tcl

3togo, 2022-01-21:
With reference to the comments on PYNQ/boards/ZCU104/base/base.tcl, it should support 2020.2
 # 1.00a po  07/15/2017 initial release
 # 2.04  yrq 01/17/2019 update to 2018.3
 # 2.5   yrq 08/22/2019 update to 2019.1
 # 2.6   yrq 11/06/2019 update to 2020.1
 # 2.70  mr  05/17/2021 update to 2020.2 

However, when I use vivado 2022.2 to do "tools -> run tcl scripts -> base.tcl ", I got the following errors.
`ERROR: [BD 5-106] Arguments to the connect_bd_intf_net command cannot be empty.
ERROR: [Common 17-39] 'connect_bd_intf_net' failed due to earlier errors.

    while executing
"connect_bd_intf_net -intf_net axi_interconnect_M08_AXI [get_bd_intf_pins s_axi_control] [get_bd_intf_pins color_convert/s_axi_control]"
    (procedure "create_hier_cell_hdmi_in" line 86)
    invoked from within
"create_hier_cell_hdmi_in $hier_obj hdmi_in"
    (procedure "create_hier_cell_video" line 99)
    invoked from within
"create_hier_cell_video [current_bd_instance .] video"
    (procedure "create_root_design" line 188)
    invoked from within
"create_root_design """
    (file "/sf_dir/PYNQ/boards/ZCU104/base/base.tcl" line 3470)
`    

How to fix it ?

EmbeddedTec, 2022-01-21:
have you checked, that the HLS components are generated successfully? there is currently a year date bug in vitis HLS that leads to errors during export: 
https://support.xilinx.com/s/question/0D52E00006uxy49SAA/vivado-fails-to-export-ips-with-the-error-message-bad-lexical-cast-source-type-value-could-not-be-interpreted-as-target?language=en_US

mariodruiz, 2022-01-21:
Have you applied this patch your installation? https://support.xilinx.com/s/article/76960?language=en_US

We have an active forum were we answer PYNQ related questions here https://discuss.pynq.io/

3togo, 2022-01-22:
thank you to @mariodruiz and @EmbeddedTec . I confirmed that it is due to the y2k22 bug. 

As I need to patch more than one version of vivado, I have written a bash script to do it.  The source code of this script is attached below. 

Hope someone might find it useful.

```
eli@jc-acer:/tools/Xilinx/y2k22_patch$ cat runme.sh 
#! /bin/bash
cd $(dirname "$0")
patch_prg="patch.py"
[[ ! -f $patch_prg ]] && echo "$patch_prg not found" && exit 1
vivado_dirs=""
for mdir in tools opt; do
    mdir="/$mdir/Xilinx/Vivado"
    [[ ! -d $mdir ]] && continue
    vivado_dirs+="$(find $mdir -mindepth 1 -maxdepth 1 -type d) "
done

for vivado_dir in $vivado_dirs; do
    ver=${vivado_dir##*/}
    python_dir="$(find $vivado_dir/tps/lnx64/ -maxdepth 1 -name "python*")"
    echo $python_dir
    python_bins="$(find $python_dir/bin -maxdepth 1 -regex '.*/python[0-9\.]*')"
	python_bins=(${python_bins[@]})
	python_bin="${python_bins[0]}"
    echo "[$python_bin]"
    cmds=("export LD_LIBRARY_PATH=${python_dir}/lib/")
    cmds+=("${python_bin} patch.py")
    for cmd in "${cmds[@]}"; do
        echo "$cmd"
        eval $cmd
    done
done

```

[y2k22_patch.tar.gz](https://github.com/Xilinx/PYNQ/files/7918420/y2k22_patch.tar.gz)

closed 2022-01-22 by 3togo