Xilinx/PYNQ #1335: Build fail using 2021.2

duaneellissd, 2022-02-06:
Step 1: Git clone
Step 2: Execute "settings.sh" in the Vititas directory and the Vivado directory
Step 3: Run "build.sh" in the top level, 

Eventually, it states:
```
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 7.99 seconds. CPU system time: 1.13 seconds. Elapsed time: 8.26 seconds; current allocated memory: -874.645 MB.
INFO: [HLS 200-1510] Running: export_design -format ip_catalog -description Controller for the trace analyzer with 32-bit data -version 1.4 -display_name Trace Analyzer Controller with 32 Bits Data 
INFO: [IMPL 213-8] Exporting RTL as a Vivado IP.

****** Vivado v2021.2 (64-bit)
  **** SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
  **** IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source run_ippack.tcl -notrace
ERROR: '2202041517' is an invalid argument. Please specify an integer value.
    while executing
"rdi::set_property core_revision 2202041517 {component component_1}"
    invoked from within
"set_property core_revision $Revision $core"
    (file "run_ippack.tcl" line 979)
INFO: [Common 17-206] Exiting Vivado at Fri Feb  4 15:17:19 2022...
ERROR: [IMPL 213-28] Failed to generate IP.
INFO: [HLS 200-111] Finished Command export_design CPU user time: 13.25 seconds. CPU system time: 1.29 seconds. Elapsed time: 16.61 seconds; current allocated memory: 6.570 MB.
command 'ap_source' returned error code
    while executing
"source trace_cntrl_32/script.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total CPU user time: 22.34 seconds. Total CPU system time: 2.97 seconds. Total elapsed time: 26.08 seconds; peak allocated memory: 1.033 GB.
INFO: [Common 17-206] Exiting vitis_hls at Fri Feb  4 15:17:22 2022...
child process exited abnormally
INFO: [Common 17-206] Exiting Vivado at Fri Feb  4 15:17:22 2022...
makefile:10: recipe for target 'hls_ip' failed
make: *** [hls_ip] Error 1
duane@ubuntu:~/pete/PYNQ$ 
```

duaneellissd, 2022-02-06:
NOTE: going back to v 2019.2 does not solve the problem, same basic error but slightly different wording.

This patch from Xilinx claims it would help, it does not.

https://support.xilinx.com/s/article/76960?language=en_US

schelleg, 2022-02-06:
PYNQ v2.7 is built using 2020.2 tools ... we typically call build.sh through the sdbuild/Makefile, which would check your env and throw an error on Vivado version mismatch.

nick-petrovsky, 2022-05-10:
> NOTE: going back to v 2019.2 does not solve the problem, same basic error but slightly different wording.
> 
> This patch from Xilinx claims it would help, it does not.
> 
> https://support.xilinx.com/s/article/76960?language=en_US

You have to patch Xilinx tools. Follow this ref: https://support.xilinx.com/s/article/76960?language=en_US

julianctrt, 2022-06-16:
Hi,
i'm also using this version and don't would like to install another old version.

any workaround to build with Vivado 2021.2?.

The last pynq is the 2.8 and also is for the Vivado 2020.2. why?. we are in 2022.1 version and increasing...

mariodruiz, 2022-10-17:
PYNQ v3.0 has just been released. This version uses Vivado 2022.1.

closed 2022-10-17 by mariodruiz