Xilinx/PYNQ #1344: XRT Device fails with newer XRT versions

quetric, 2022-03-04:
* PYNQ version: 2.7.0 (pip installed)
* Board name: Alveo (any)

* Description: XRT version identification fails with newer XRT versions. This is because xrt_device.py at line 88 tries to do `xbutil dump` to get the version, but that command no longer exists. 

* Suggested fix: The function should run `xbutil --version` instead (or as a backup if `xbutil dump` fails), and parse the first line of output e.g. `output.stdout.splitlines()[0].split(": ")[1].split(".")`

mariodruiz, 2022-03-04:
Hi @quetric, we are working on a solution. 
There is a PR already opened for this here https://github.com/Xilinx/PYNQ/pull/1343 

Another workaround for now is to use the `--legacy` switch https://discuss.pynq.io/t/pynq-on-alveo-and-xrt-2021-2/3568/2?u=marioruiz

quetric, 2022-03-04:
Thank you @mariodruiz, work-around noted.

mariodruiz, 2022-10-17:
Fixed in v3.0

PR #1343 (2022-06-15): Use the system-wide files to get XRT version
Instead of running `xbutil` use the already present files in the system to get the XRT version. This seems to be a cleaner way of doing this.

This solves interoperability with XRT > 2.12

Fix #1344
closed 2022-10-17 by mariodruiz