Xilinx/PYNQ #1348: PYNQ Z1 Sample Rate

ELKTKJ, 2022-03-08:
- PYNQ version : v2.1
- Board name : Pynq-Z1

The sample rate is too high for me, can i record with a lower sample rate. Or can i speed up the process of PDM to PCM.
Looking forward to your replay. Thank you.

mariodruiz, 2022-03-08:
Hi, please post these questions on the pynq forums. https://discuss.pynq.io/c/support/30

ELKTKJ, 2022-03-08:
OK

ELKTKJ, 2022-03-08:
Sorry, In  pynq forums, it shows Access Denied when I creat a new topic.

mariodruiz, 2022-03-08:
Is your username `liuxiaocs`? If so, please try to post again

ELKTKJ, 2022-03-09:
Yes, I'll try to post again later, Thank you!

ELKTKJ, 2022-03-09:
Sorry, the same problem occurs when I try to create topic.
[image]

[image]

ELKTKJ, 2022-03-09:
[image]

mariodruiz, 2022-03-09:
Please try again, you were silence because you typed too fast.

ELKTKJ, 2022-03-09:
It' works, Thank you!