Xilinx/PYNQ #1366: SD image - sdbuild make fail about petalinux bsp

Kgfu, 2022-05-23:
I'm following the guide to build the image in VM through vagrant:
https://pynq.readthedocs.io/en/v2.7.0/pynq_sd_card.html
In the step:
cd <PYNQ repository>/sdbuild/
make

I got following error message seems related to the peta-linux BSP:
[image]

I confirm the version of Vitis, Vivado, and Petalinux are the same. (2020.2)
Does it possible cause by the petalinux installation is not completed?

skalade, 2022-05-23:
Hi there,

Are you using Ubuntu 18.04? And have you ran the sdbuild/scripts/setup_host.sh script? There's usually logs in the sdbuild/build directory, you can find it by doing `find . -name "build.log"`, which might give more hints as to what is going on.

I recommend you create a post on the [pynq support forum](https://discuss.pynq.io) with issues like this -- you are more likely to receive support. Just please make sure to describe what steps you have taken to debug the issue, and give as much information about your setup as possible.

Thanks
Shawn

Kgfu, 2022-05-24:
Hi skalade, 
I'm using the vagrant to create the VM by default, so the OS is Ubuntu 18.04.
The setup_host script is executed during the command: vagrant up that creating VM and initial it.
I'm not find other error message or warming during the VM initialization, so I believe the setup_host.sh script should not have problem.

Got it. I will follow your suggestion to support forum. Thanks~

Idosegevv, 2022-11-22:
@Kgfu Try to create a VM with ubuntu 20.04. install all Linux dependencies(also with the help of the setup script). And then try.

skalade, 2024-05-01:
Closing due to lack of activity. If this is still an issue re-open it as a [discuss.pynq.io](https://discuss.pynq.io/) post.

closed 2024-05-01 by skalade