Xilinx/PYNQ #1392: _SGDMAChannel.transfer() ignores start offset

sebastianpetersFAU, 2022-10-04:
I just noticed that in [dma.py](https://github.com/Xilinx/PYNQ/blob/image_v2.7/pynq/lib/dma.py#L347), when using the `_SGDMAChannel.transfer(array, start, nbytes)` function, the `start` parameter (offset into array) is ignored in lines [427](https://github.com/Xilinx/PYNQ/blob/image_v2.7/pynq/lib/dma.py#L427) and [429](https://github.com/Xilinx/PYNQ/blob/image_v2.7/pynq/lib/dma.py#L429) when the descriptors are filled out.

This leads to the data ending up at the wrong address as if `start` was always set to `0`.

In line [382](https://github.com/Xilinx/PYNQ/blob/image_v2.7/pynq/lib/dma.py#L382) the `start` parameter is used as expected `... (array.physical_address + start) ...` when checking for alignment,

-----

**Suggestion:** Add `+ start` in line [427](https://github.com/Xilinx/PYNQ/blob/image_v2.7/pynq/lib/dma.py#L427) (and [429](https://github.com/Xilinx/PYNQ/blob/image_v2.7/pynq/lib/dma.py#L429)):

old: `(array.physical_address + (i * blk_size)) & 0xffffffff`
new: `(array.physical_address + start + (i * blk_size)) & 0xffffffff`

mariodruiz, 2022-10-17:
Hi @sebastianpetersFAU,

The version 3.0 has just been released.

However, I suggest you create a Pull Request with these changes, and they will be reviewed.

Mario

sebastianpetersFAU, 2022-10-24:
Hi @mariodruiz 

I have created a pull request (https://github.com/Xilinx/PYNQ/pull/1401) since this issue exists in version 3.0 as well.

Sebastian

closed 2024-05-01 by skalade