Xilinx/PYNQ #1405: Failed to build PYNQ image file for ZCU102

CharlesChen1024, 2022-10-25:
Hi, 
I try to run "make boot_files BOARDS=ZCU102" to get pynq image file for zcu102. But when I run this command, it show some error message like the figure.
[image]
[image]
How to solve them?

mariodruiz, 2022-10-25:
Hi @CharlesChen1024,

Please post your questions on the pynq.io forums https://discuss.pynq.io/

schelleg, 2024-05-01:
Closing due to lack of activity. If this is still an issue re-open it as a [[discuss.pynq.io](http://discuss.pynq.io/)](https://discuss.pynq.io/) post.

closed 2024-05-01 by schelleg