Xilinx/PYNQ #1407: Undeclared class members on line 548 (/lib/video/dma.py)

PREDATORNaruz, 2022-11-19:
* PYNQ version : v3.0.0 and v3.0.1
* Board name : Pynq Z1
* Pull request type : bug fix 
* Pull request number:
* Description: undeclared members, self.s2mm_introut and self.mm2s_introut on __init__ line 548 (/lib/video/dma.py)  
*                           this will issue a failure "unrecognized self.s2mm_introut and self.mm2s on line 580 and 582

mariodruiz, 2022-11-21:
@PREDATORNaruz,

I don't think this is a bug. Please post your question in the [https://discuss.pynq.io/](https://discuss.pynq.io/) forum. 

I assume you have not connected the interrupts as described in the documentation https://pynq.readthedocs.io/en/latest/pynq_libraries/interrupt.html 

Mario

PREDATORNaruz, 2022-11-22:
@mariodruiz  
You're right, on my block design the s2mm and mm2s interrupts of the VMDA goes to the concat block and directly the PS....
However, after initializing mm2s and s2mm in dma.py (just adding self. = '' to both interrupts) everything worked fine...

here's a code snipped 
*                  vdma.write(0x00,0x04)
while vdma.read(0x00) & 0x4 == 4: 
    pass

vdma.write(0x30,0x04)
while vdma.read(0x30) & 0x4 == 4: 
    pass

vdma.write(0x00,0x93)
vdma.write(0x5C,addr_inp_buff)
vdma.write(0x58,widht*3)
vdma.write(0x54,widht*3)
vdma.write(0x50,hight)

vdma.write(0x30,0x93)
vdma.write(0xAC,addr_output_buff)
vdma.write(0xA8,widht)
vdma.write(0xA4,widht)
vdma.write(0xA0,hight)

while vdma.register_map.S2MM_VDMASR.Halted != 1:
    pass

PREDATORNaruz, 2022-11-22:
@mariodruiz
You're right, on my block design the s2mm and mm2s interrupts of the VMDA goes to the concat block and directly the PS....
However, after initializing mm2s and s2mm in dma.py (just adding self. = '' to both interrupts) everything worked fine...

here's a code snipped

             vdma.write(0x00,0x04)
             while vdma.read(0x00) & 0x4 == 4:
                  pass

             vdma.write(0x30,0x04)
             while vdma.read(0x30) & 0x4 == 4:
                  pass

             vdma.write(0x00,0x93)
             vdma.write(0x5C,addr_inp_buff)
             vdma.write(0x58,widht3)
             vdma.write(0x54,widht3)
             vdma.write(0x50,hight)

             vdma.write(0x30,0x93)
             vdma.write(0xAC,addr_output_buff)
             vdma.write(0xA8,widht)
             vdma.write(0xA4,widht)
             vdma.write(0xA0,hight)

             while vdma.register_map.S2MM_VDMASR.Halted != 1:
                 pass

mariodruiz, 2022-11-22:
> Do you recommend going through Axi Interrupt Controller like the guide says ?

Yes, for more than 1 interrupt, using the interrupt controller is the only way that PYNQ will associate the interrupts.

closed 2022-11-22 by mariodruiz