Xilinx/PYNQ #1408: Cannot load Overlay when processing_system7 IP is inside hierarchy

dspsandbox, 2022-11-25:
After migrating to PYNQ 3.0.0,  _pynq.Overlay("bistream_name.bit")_ returns a key error when the underlying FPGA designs has the processing system IP located within a hierarchy. In the traceback below you can see the error when placing the processing system IP (name: "processing_system7_0" ) within a hierarchy (name: "test").

After some investigations it seems that the error is caused by the _get_processing_systems()_ (part of PYNQ-Metadata). This functions returns a dictionary with a key of the form  "test_processing_system7_0" which probably should be  "test/processing_system7_0".

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Input In [1], in <cell line: 24>()
     13 bh._clear_cache()
     15 # #Patch get_processing_systems() method
     16 # def get_processing_systems_patch(self):
     17 #     for core in self.blocks.values():
   (...)
     21 # pynq.metadata.runtime_metadata_parser.Module.get_processing_systems = get_processing_systems_patch
     22 #############################################
---> 24 ol = pynq.Overlay("dds.bit")
     25 buf_0 = pynq.allocate(shape=(1000000,), dtype=np.int16)
     26 buf_1 = pynq.allocate(shape=(1000000,), dtype=np.int16)

File /usr/local/share/pynq-venv/lib/python3.10/site-packages/pynq/overlay.py:348, in Overlay.__init__(self, bitfile_name, dtbo, download, ignore_version, device, gen_cache)
    345     self.systemgraph = None
    347 if download:
--> 348     self.download()
    349 else:
    350     if gen_cache:

File /usr/local/share/pynq-venv/lib/python3.10/site-packages/pynq/overlay.py:428, in Overlay.download(self, dtbo)
    425         else:
    426             Clocks.set_pl_clk(i)
--> 428 super().download(self.parser)
    429 if dtbo:
    430     super().insert_dtbo(dtbo)

File /usr/local/share/pynq-venv/lib/python3.10/site-packages/pynq/bitstream.py:168, in Bitstream.download(self, parser)
    151 def download(self, parser=None):
    152     """Download the bitstream onto PL and update PL information.
    153 
    154     If device tree blob has been specified during initialization, this
   (...)
    166 
    167     """
--> 168     self.device.download(self, parser)

File /usr/local/share/pynq-venv/lib/python3.10/site-packages/pynq/pl_server/embedded_device.py:689, in EmbeddedDevice.download(self, bitstream, parser)
    686 with open(self.BS_FPGA_MAN, "w") as fd:
    687     fd.write(bitstream.binfile_name)
--> 689 self.set_axi_port_width(parser)
    691 self._xrt_download(parser.xclbin_data)
    692 super().post_download(bitstream, parser, self.name)

File /usr/local/share/pynq-venv/lib/python3.10/site-packages/pynq/pl_server/embedded_device.py:647, in EmbeddedDevice.set_axi_port_width(self, parser)
    644 if not hasattr(parser, "ps_name"):
    645     # Setting port widths not supported for xclbin-only designs
    646     return
--> 647 parameter_dict = parser.ip_dict[parser.ps_name]["parameters"]
    648 if parser.family_ps == "zynq_ultra_ps_e":
    649     for para in ZU_FPD_SLCR_REG:

KeyError: 'test_processing_system7_0'

STFleming, 2022-11-25:
Hi @dspsandbox,

Thanks for reporting this, I'll look into recreating the bug and getting a PR up for the fix.

Thanks,
Shane

STFleming, 2022-11-25:
Hi @dspsandbox,

I think I have a fix for this up if you wouldn't mind trying it out. On your board, in a jupyter lab terminal, can you run the following command to install a version of PYNQ with my fix included:
```
python3 -m pip install git+https://github.com/STFleming/PYNQ.git@ps_hier_fix --no-build-isolation
``` 

Then try refreshing the notebook and reloading the overlay, hopefully it should work as expected. 

[image]

If that works for you I can do some more testing and get a PR up.

dspsandbox, 2022-11-25:
Hi Shane,
the PYNQ update resolved the hierarchy problem. 
Thanks again for pushing this out so fast,
Pau

STFleming, 2022-11-26:
Hi Pau,

No worries at all. I'll do some more testing on the fix, I think it looks okay, and get a proper PR up. Thanks again for reporting it.

Shane

mariodruiz, 2025-04-07:
Looks like this is already merged in the image_v3.1 branch and it is has been tested.

closed 2026-09-02 by jogomojo