Xilinx/PYNQ #1411: Invalid boundary value in pmod_pwm.py

haipnh, 2023-01-17:
Hi,

I found an invalid boundary value in [pynq/lib/pmod/pmod_pwm.py](https://github.com/Xilinx/PYNQ/blob/master/pynq/lib/pmod/pmod_pwm.py).

```python3
def generate(self, period, duty_cycle):
        # ...
        if period not in range(1, 65536):
            raise ValueError("Valid tone period is between 1 and 65536.")
        if duty_cycle not in range(1, 99):
            raise ValueError("Valid duty cycle is between 1 and 99.")
       # ...
```

On Jupyter Lab, it fails if we execute:
```python3
pwm_1.generate(65536,25)
```

Error message:
```
/usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq/lib/pmod/pmod_pwm.py in generate(self, period, duty_cycle)
     95         """
     96         if period not in range(1, 65536):
---> 97             raise ValueError("Valid tone period is between 1 and 65536.")
     98         if duty_cycle not in range(1, 99):
     99             raise ValueError("Valid duty cycle is between 1 and 99.")

ValueError: Valid tone period is between 1 and 65536.
```

Another failed point:
```python3
pwm_1.generate(10,99)
```
Caused:
```
/usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq/lib/pmod/pmod_pwm.py in generate(self, period, duty_cycle)
     97             raise ValueError("Valid tone period is between 1 and 65536.")
     98         if duty_cycle not in range(1, 99):
---> 99             raise ValueError("Valid duty cycle is between 1 and 99.")
    100 
    101         self.microblaze.write_mailbox(0, [period, duty_cycle])

ValueError: Valid duty cycle is between 1 and 99.
```

Best regards,
Hai Pham

mariodruiz, 2023-01-18:
Hi @haipnh,

We appreciate if you can contribute the solution to this issue.

Mario

skalade, 2024-05-01:
Looks like working as intended, python `range` does not include the last number.

closed 2024-05-01 by skalade