Xilinx/PYNQ #1425: PYNQ image for ZCU106

ptoupas, 2023-06-20:
As far as I understand and from a quick search that I have conducted I can not find any PYNQ image for the ZCU106 Xilinx board.

I would like to ask:
1. is there is any plans to support this specific board?
2. is the any way for me to generate such an image and if so could you point me to some directions towards building it?

Thank you very much in advance.

ptoupas, 2023-06-21:
I would like to give an update regarding the previous state of the issue.

I have managed to generate the PYNQ image for the ZCU106 board following the steps below.

1. Install the Vivado/Vitis/Petalinux tools for the version of the PYNQ you are interested (in my case the combination was 2022.1 for Vivado/Vitis/Petalinux and 3.0.1 for the PYNQ)
2. Install all the [prerequisites and necessary libraries](https://support.xilinx.com/s/article/000033799?language=en_US) for Petalinux
3. Download the ZCU106 BSP for 2022.1 version of the Xilinx tools from [here](https://www.xilinx.com/member/forms/download/xef.html?filename=xilinx-zcu106-v2022.1-04191534.bsp)
4. Download the [PYNQ rootfs aarch64 v3.0.1](https://bit.ly/pynq_aarch64_v3_0_1) and [Prebuilt PYNQ source distribution binary v3.0.1](https://bit.ly/pynq_sdist_v3_0_1)
5. Clone the PYNQ repository
6. Execute the following command `<PYNQ repository>/sdbuild/scripts/setup_host.sh`
7. Copy the two PYNQ files downloaded in step 4 into the PYNQ repository:
`cp pynq_rootfs.<arm|aarch64>.tar.gz <PYNQ repository>/sdbuild/prebuilt/pynq_rootfs.<arm|aarch64>.tar.gz` and
`cp pynq-<version>.tar.gz <PYNQ repository>/sdbuild/prebuilt/pynq_sdist.tar.gz`
8. Source your Vitis and Petalinux like this:
`source <path-to-vitis>/Vitis/2022.1/settings64.sh` and
`source <path-to-petalinux>/petalinux-2022.1-final/settings.sh`
9. Change your terminal into bash by executing the command `sudo dpkg-reconfigure dash` and pressing no when prompt.
10. Navigate to `<PYNQ repository>/sdbuild/scripts/` and execute the following command 
`./image_from_prebuilt.sh ZCU106 <BSP stored repository>/xilinx-zcu106-v2022.1-04191534.bsp aarch64 <PYNQ repository>/sdbuild/prebuilt/pynq_rootfs.aarch64.tar.gz`
11. Relax and wait for the tool to generate the final image than you can use for your SD card.

Most of these steps and instructions can be found either on [PYNQ github issues](https://github.com/Xilinx/PYNQ/issues) page or on the [official PYNQ forum](https://discuss.pynq.io/) page. However I couldn't find a complete and end to end guide for this specific platform (or some other non-supported one) so I have gathered the steps that I have made and provided here some more comprehensive and end to end instructions.

I hope the information I provide can be helpful and save time to other people.

closed 2023-06-21 by ptoupas

ColeFrancis, 2025-04-20:
### Issue with build script

I go through all the steps but when I run the image_from_prebuild.sh as described I get the error:

`` Copying files into the device: __populate_fs: Could not allocate block in ext2 fimkfs.ext4: Could not allocate block in ext2 filesystem while writing file massif-arm64-linun ``

``mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system ``

`` ERROR: Failed to create wic image ``

I believe that the script is not allocating enough memory for the massif-arm64-linux file while creating the image.