Xilinx/PYNQ #1429: DebugBridge fails to connect

jacopoabramo, 2023-08-22:
Greetings,

I tried using the `DebugBridge` class together with the debug bridge IP in my design but could not make it work. I documented the problem in [this post](https://discuss.pynq.io/t/using-the-debugbridge-to-visualize-an-ila-in-vivado/5977). Does the original author of this class have some idea on what am I doing wrong?

* PYNQ version 3.0.1
* board PYNQ-ZU
* IP generated with Vivado 2023.1

Thanks in advance.

Best,

Jacopo

fpjentzsch, 2024-03-31:
Hi,

looks like a simple bug to me:
The DebugBridge constructor checks if the `description` is of type `ReprDict` ([this line](https://github.com/Xilinx/PYNQ/blob/93ddd21ab623883590a8c8b07f0b157b2855da4b/pynq/lib/debugbridge.py#L209)) while other drivers, such as the DMA, check for `ReprDict` or `dict` ([here](https://github.com/Xilinx/PYNQ/blob/93ddd21ab623883590a8c8b07f0b157b2855da4b/pynq/lib/dma.py#L587C37-L587C53)).

I was able to work around this using:
```
from pynq.lib.debugbridge import DebugBridge
from pynqutils.runtime.repr_dict import ReprDict

bridge_desc = ReprDict(ol.ip_dict["debug_bridge_0"]) # ol is the Overlay
bridge = DebugBridge(bridge_desc)
bridge.start_xvc_server(bufferLen=4096, serverAddress="0.0.0.0", serverPort=2542, reconnect=True, verbose=True)
```

fpjentzsch, 2024-04-15:
> That's a good point @fpjentzsch. That check is not really necessary, although it can be fixed as you suggest. Could you do a PR?

Sure: https://github.com/Xilinx/PYNQ/pull/1445

closed 2024-04-16 by schelleg