Xilinx/PYNQ #1456: Numpy 2.0 breaks pynq.allocate on XRT Platforms

drichmond, 2024-07-23:
(I _think_ I have it diagnosed)

Numpy added a device getter to ndarray in Version 2.0. ([doc](https://numpy.org/doc/stable/reference/generated/numpy.ndarray.device.html), [src](https://github.com/numpy/numpy/blob/1d49c7f7ff527c696fc26ab2278ad51632a66660/numpy/_core/src/multiarray/getset.c#L969))

When allocating a buffer on an XRT platform, the default target works just fine because I don't believe it touches the device field of PynqBuffer. 

However, when a non-default target for allocation is provided, it fails with the following exception:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[4], line 1
----> 1 input = pynq.allocate(1024, 'u4', target=ol.plram)
      2 input[:] = np.ones(1024)

File ~/.local/lib/python3.10/site-packages/pynq/buffer.py:158, in allocate(shape, dtype, target, **kwargs)
    156 if target is None:
    157     target = Device.active_device
--> 158 return target.allocate(shape, dtype, **kwargs)

File ~/.local/lib/python3.10/site-packages/pynq/pl_server/xrt_device.py:156, in XrtMemory.allocate(self, shape, dtype, **kwargs)
    145 def allocate(self, shape, dtype, **kwargs):
    146     """Create a new  buffer in the memory bank
    147
    148     Parameters
   (...)
    154
    155     """
--> 156     buf = _xrt_allocate(shape, dtype, self.device, self.idx, **kwargs)
    157     buf.memory = self
    158     return buf

File ~/.local/lib/python3.10/site-packages/pynq/pl_server/xrt_device.py:107, in _xrt_allocate(shape, dtype, device, memidx, cacheable, pointer, cache)
    105     buf = device.map_bo(bo)
    106     device_address = device.get_device_address(bo)
--> 107 ar = PynqBuffer(
    108     shape,
    109     dtype,
    110     bo=bo,
    111     device=device,
    112     buffer=buf,
    113     device_address=device_address,
    114     coherent=False,
    115 )
    116 if pointer is not None:
    117     weakref.finalize(buf, _free_bo, device, bo, ar.virtual_address, ar.nbytes)

File ~/.local/lib/python3.10/site-packages/pynq/buffer.py:45, in PynqBuffer.__new__(cls, device, device_address, bo, coherent, *args, **kwargs)
     43 self.coherent = coherent
     44 self.bo = bo
---> 45 self.device = device
     46 self.offset = 0
     47 self.freed = False

AttributeError: attribute 'device' of 'numpy.ndarray' objects is not writable
```

If I edit buffer.py and change the name of the device field to ndevice, no exception is thrown, and my code works just fine.

drichmond, 2024-07-23:
Forgot to mention: Downgrading to Numpy 1.26.4 also fixes the issue.

mariodruiz, 2024-07-24:
Hi Dustin,

Thank you for reporting this, We're tracking this, see https://github.com/Xilinx/PYNQ/pull/1453

drichmond, 2024-07-24:
Race condition! Do you want me to close?

mariodruiz, 2024-07-24:
If it is resolved, please close.

closed 2024-07-25 by drichmond