Xilinx/PYNQ #1463: CRITICAL WARNING: [Timing 38-249] Generated clock base_i/video/hdmi_out/frontend/rgb2dvi_0/U0/SerialClk has no logical paths from master clock axi_dynclk_0_PXL_CLK_O

dzung-hoang, 2024-11-28:
Regenerating base overlay for Pynq-Z1 results in following critical warning.
```
CRITICAL WARNING: [Timing 38-249] Generated clock base_i/video/hdmi_out/frontend/rgb2dvi_0/U0/SerialClk has no logical paths from master clock axi_dynclk_0_PXL_CLK_O.
Resolution: Review the path between the master clock and the generated clock with the schematic viewer and correct the -source option. If it is correct and the master clock does not have a timing path to the generated clock, define the generated clock as a primary clock by using create_clock.
```
The root cause is this unnecessary line in `./boards/ip/rgb2dvi_v1_2/src/rgb2dvi_clocks.xdc`.

```
### Clock constraints ###
create_generated_clock -source [get_ports PixelClk] -multiply_by 5 [get_ports SerialClk]
```

Commenting out the create_generate_clock line fixes the critical warning.

mariodruiz, 2025-04-07:
Hi @dzung-hoang,

Can you please open a PR with this fix against image_v3.1?

dzung-hoang, 2025-04-08:
Done.
Regards,Dzung