Xilinx/PYNQ #1479: Why is typing_extensions pinned to 4.3.0?

zzzhhh2025, 2025-05-20:
[`sdbuild/packages/python_packages_jammy/requirements.txt`](https://github.com/Xilinx/PYNQ/blob/a056b8455f80a145839177102288e1f1d2b8ebe3/sdbuild/packages/python_packages_jammy/requirements.txt) requires that the version of typing_extensions be pinned to 4.3.0. This would generate the following error during installation:
```
Installing collected packages: typing_extensions
  Attempting uninstall: typing_extensions
    Found existing installation: typing_extensions 4.13.2
    Uninstalling typing_extensions-4.13.2:
      Successfully uninstalled typing_extensions-4.13.2
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
exceptiongroup 1.3.0 requires typing-extensions>=4.6.0; python_version < "3.13", but you have typing-extensions 4.3.0 which is incompatible.
Successfully installed typing_extensions-4.3.0
```
So, is there any special reason to pin typing_extensions to such a low version of 4.3.0, when a higher version of 4.13.2 exists and has been installed in Ubuntu 22.04? If there is no special reason of dependency, can any developer of PYNQ change the requirement from `typing_extensions==4.3.0` to `typing-extensions>=4.6.0` as indicated by the error message? Thanks.

mariodruiz, 2025-05-20:
Hi @zzzhhh2025,

We verify the `pynq` environment with a fixed version of the packages. You can try to update the package, but a different version is something we have not verified

zzzhhh2025, 2025-05-21:
> Hi [@zzzhhh2025](https://github.com/zzzhhh2025),
> 
> We verify the `pynq` environment with a fixed version of the packages. You can try to update the package, but a different version is something we have not verified

First, it is a convention to develop package/library to be backward compatible. So, there is no need to test all versions.

Second, the ultimate purpose of sticking to a fixed version (and verifying based on it) is to avoid a bug. But as I said,  `typing_extensions==4.3.0` will cause an error in installation, so your response should be trying to fix this error based on my post, but not trying to justify you sticking to a buggy version and asking users to fix by themselves. If you are an Xilinx employee, your attitude to pass the buck is enough to lose your job.

closed 2025-09-24 by jogomojo